ROBOTIS-GIT/turtlebot3_autorace #5: after git clone the repository from Turtlebot3, when i cd catkinws and cmake, screen hang

timlo94, 2018-05-14:
first i go cd catkinws/src and i git clone this repository, afterward i cmake and the screen hang directly at 93%. i had perform apt update,upgrade,remove . Even after i reboot the raspberrypi3, i can still run the node with some error. like lane_detection no PYTHON EXTENSION. should be lane_detection.py!!!!

ROBOTIS-Leon, 2018-05-16:
@timlo94 

Sorry,
first of all, I should change all the python files' name to nonextension.
I will update after the modification.

ROBOTIS-Leon, 2018-05-16:
@timlo94 

It must be fixed now. 
Let's go ahead

timlo94, 2018-05-16:
thanks for your hard work. i will try it tomorrow morning.

robotpilot, 2018-05-20:
Thanks, please let us know if you have any other problems.

timlo94, 2018-05-21:
After redownload and compile, it was successful. But after run the intrinsic camera calibration, the displau rqt gui dimmed and show nothing, and the lainch node cannot be runned due to CVbridge error

robotpilot, 2018-05-21:
@ROBOTIS-Leon , 
We need your guidance about camera calibration and dependency package.

robotpilot, 2018-05-27:
@kijongGil 
Has this problem been resolved? 
Please take a look at this issue and add a detailed description of it to e-manual.
http://emanual.robotis.com/docs/en/platform/turtlebot3/autonomous_driving/#autonomous-driving

timlo94, 2018-05-28:
Not yet solved. It seems like module object has no attribute, CVbridge error simply cannot solve just by downloaded respective dependencies or update/install relevant packages. By the way i am using raspicam, i am not sure whether your cv using usbcam or not.

And also ya, can i know is it other peoples who downloaded this repository also faced same situation? Or developer has tested the code several times before release?.

Thanks a lot for developer senior research engineer on this matter. Your problem solving skills is appreciated .

ROBOTIS-Leon, 2018-05-28:
@timlo94

Hi and sorry for the late reply.
I couldn’t notice that there was some extra questions, since I was in the vacation.

the attached shows something weird; if you ran the catkin make, the opencv bridge must be installed by the joined dependent package. so anyway the error means the cv bridge couldnt be found via python in your computer. 
try sudo apt-get install ros-kinetic-cv-bridge.

readme.md shows how and what kinda camera should be used for this code. See the description carefully.

There are number of users who ran the code already in their robots. I personally think that you are able to handle

timlo94, 2018-05-29:
[image]
[image]

timlo94, 2018-05-29:
[image]
[image]

timlo94, 2018-05-29:
My cv bridge is up to date and i had follow readme,just my camera used is normal raspberry pi camera( from official turtlebot)

timlo94, 2018-05-30:
Thanks for updating the repository.

kijongGil, 2018-05-31:
Hi, @timlo94
How are you doing?

closed 2018-06-06 by robotpilot