ROBOTIS-GIT/turtlebot3_autorace #12: Warning at step 3: Run AutoRace launch

tranvannhancu, 2018-07-02:
**I got warning at step 3 after entering:** roslaunch turtlebot3_autorace_camera turtlebot3_autorace_intrinsic_camera_calibration.launch 

**Here is what appeared afterward:**
 ... logging to /home/nhan/.ros/log/848851d0-7e04-11e8-a6d2-ac7ba19ca973/roslaunch-nhan-zbook-6006.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://nhan-zbook:37727/

SUMMARY
========

PARAMETERS
 * /camera/image_proc/queue_size: 20
 * /rosdistro: kinetic
 * /rosversion: 1.12.13

NODES
  /camera/
    image_proc (image_proc/image_proc)
  /
    republish (image_transport/republish)

ROS_MASTER_URI=http://localhost:11311

process[republish-1]: started with pid [6023]
process[camera/image_proc-2]: started with pid [6024]
[ WARN] [1530542037.119968670]: The input topic '/camera/camera_info' is not yet advertised

**This seems to be the source of error since when I finish the rest of tutorial, the turtlebot is not running autonomously at all.**

kijongGil, 2018-07-03:
Hi :) tranvannhancu
I am sorry for your inconvenience.
This problem is the difference between the origin urdf and gazebo urdf topic names.
We will solve and update it. But If you want to try it now, change to topic name in turtlebot3/turtlebot3_description/urdf/turtlebot3_burger_for_autorace.gazebo.xacro file.

https://github.com/ROBOTIS-GIT/turtlebot3/blob/c2e272faf915e6cb416d50fa16ef971c6f402d9a/turtlebot3_description/urdf/turtlebot3_burger_for_autorace.gazebo.xacro#L155
rgb/image_raw  -> image

https://github.com/ROBOTIS-GIT/turtlebot3/blob/c2e272faf915e6cb416d50fa16ef971c6f402d9a/turtlebot3_description/urdf/turtlebot3_burger_for_autorace.gazebo.xacro#L156
rgb/camera_info -> camera_info

Thanks,
Gilbert

tranvannhancu, 2018-07-04:
Problem solved. Thank you for quick response :)

closed 2018-07-04 by tranvannhancu

kijongGil, 2018-07-05:
I am glad that the problem has been solved. :)
If you have another problem, please reopen it.
Thanks.

ycean, 2018-07-08:
Hi,
I have been facing the same problem. I tried with the same method in the correction for the topic name in turtlebot3/turtlebot3_description/urdf/turtlebot3_burger_for_autorace.gazebo.xacro file, but the warning still came out, and the turtlebot in the gazebo doesn't move automatically.

Is there any possible error of could cause this to be happen?

kijongGil, 2018-07-08:
https://github.com/ROBOTIS-GIT/turtlebot3_autorace/issues/16