ROBOTIS-GIT/turtlebot3_autorace #15: Error while loading on PC roslaunch turtlebot3_autorace_camera turtlebot3_autorace_extrinsic_camera_calibration.launch

zujankaaa, 2018-07-05:
Hello!

Fist of all, I want to say that this is great project!

But, I have a problem, I calibrated camera following your instructions, saved the new configuration and when I run: "roslaunch turtlebot3_autorace_camera 
turtlebot3_autorace_extrinsic_camera_calibration.launch" on my PC I get the following error:

```
 error loading <rosparam> tag: 
 
 	'param' attribute must be set for non-dictionary values
 XML is <rosparam command="load" file="$(find turtlebot3_autorace_camera)/calibration/extrinsic_calibration/compensation.yaml"/>
 The traceback for the exception was written to the log file.
```

Why this happens?

zujankaaa, 2018-07-06:
Thank you for the quick reply. The name of file is "compensation.yaml" and it contains the following:

```
!!python/object/new:dynamic_reconfigure.encoding.Config
dictitems:
  ISO: 488
  awb_mode: auto
  brightness: 61
  contrast: 24
  exposureCompensation: 3
  exposure_mode: auto
  groups: !!python/object/new:dynamic_reconfigure.encoding.Config
    dictitems:
      ISO: 488
      awb_mode: auto
      brightness: 61
      contrast: 24
      exposureCompensation: 3
      exposure_mode: auto
      groups: !!python/object/new:dynamic_reconfigure.encoding.Config
        state: []
      hFlip: true
      id: 0
      name: Default
      parameters: !!python/object/new:dynamic_reconfigure.encoding.Config
        state: []
      parent: 0
      saturation: 17
      sharpness: 38
      shutterSpeed: 25000
      state: true
      type: ''
      vFlip: true
      videoStabilisation: true
      zoom: 1.0
    state: []
  hFlip: true
  saturation: 17
  sharpness: 38
  shutterSpeed: 25000
  vFlip: true
  videoStabilisation: true
  zoom: 1.0
state: []
```
I checked the path few times before, and it is the following from home directory: 

> catkin_ws/src/turtlebot3_autorace/turtlebot3_autorace_camera/calibration/extrinsic_calibration/compensation.yaml

kijongGil, 2018-07-06:
Did you save yaml file using rosrun rqt_reconfigure rqt_reconfigure?
The origin compensation.yaml file has just one parameter : clip_hist_percent
Your file is not compensation.yaml file. After calibrating extrinsic_camera using rqt_reconfigure, you have to overwrite each value.

kijongGil, 2018-07-06:
I am glad that the problem has been solved. :)
I will close this issue. If you have another problem, please reopen it.
Thanks.

closed 2018-07-06 by kijongGil

Sebastian5j, 2019-07-01:
i have the same problem, wich parameters did yo add? I dont understand the problem :c

hariharan382, 2020-10-19:
for god sake if any one solved please just dont close the issue ,just post your solution please