ROBOTIS-GIT/turtlebot3_autorace #16: Warning while launching the turtlebot3_autorace.

ycean, 2018-07-08:
I got the warning notification after I run the :roslaunch turtlebot3_gazebo turtlebot3_autorace.launch 
The warning message as shown as below:

Warning [parser.cc:778] XML Element[static], child of element[model] not defined in SDF. Ignoring[static]. You may have an incorrect SDF file, or an sdformat version that doesn't support this element.
.
.
.
[ WARN] [1531037143.322597935, 116.497000000]: DiffDrive(ns = //): missing <publishOdomTF> default is ture
[ WARN] [1531037143.323071180, 116.497000000]: DiffDrive(ns = //): missing <odometrySource> default is 1

Although the gazebo is running, but after I executed until step 3 according to the tutorial, another warning is coming out as below:

[ WARN] [1531037190.883702930, 163.390000000]: The input topic '/camera/camera_info' is not yet advertised

I did change the topic name in the file of turtlebot3/turtlebot3_description/urdf/turtlebot3_burger_gor_autorace.gazebo.xacro as what I saw in the previous issues, but the warning is still coming out, and the turtlebot in the gazebo doesn't move automatically.

kijongGil, 2018-07-08:
Hi @ycean :)
I updated solution of this issue. please remove old package and install newest packages.
You don't have to change the topic name by yourself.
You need to enter one line command.

`$ export GAZEBO_MODE=true`
The emanual has been updated as well.
http://emanual.robotis.com/docs/en/platform/turtlebot3/autonomous_driving/#autorace-with-gazebo

Thanks,
Gilbert

ycean, 2018-07-09:
Hi,  @kijongGil 
Thanks for your quick reply.
Unfortunately, the same problem still coming out.
I removed the old package and installed the latest one. Still the problem can't be solved. In addition, all my rqt_image_view also missing after I did the removal for the ros-kinetic-turtlebot3-autorace*.

Ycean.

kijongGil, 2018-07-09:
Make sure clearly removal turtlebot3 packages.
If you install turtlebot3 packages by apt-get install,  remove all packages. Then reinstall them on github.

ycean, 2018-07-09:
Hi, Gilbert

I did uninstall it both method of sudo apt-get remove [package], and sudo apt-get remove ros-kinetic-autorace* in this morning, before I replied here. 

Now I am trying to remove it again, but suddenly my ubuntu is malfunction, reinstalling everything again. Hopefully, problem can be solved.

Here I would like to reconfirm again, do I need to install the additional dependent package after I install it from github?As I notice from the 13.2.3 Tutorial :3.

Thanks again.

Ycean.

kijongGil, 2018-07-09:
Don't worry installing packages. I think that your installation is not problem. Again install follow our e-manual. I recommend to install turtlebot3 packages from github and dependent packages from apt-get.

ycean, 2018-07-09:
Thanks with noted ^^

Ycean

ycean, 2018-07-09:
Hi, 
It's working fine. Until the gate everything is good.

Thanks a lot.

ycean.