ROBOTIS-GIT/turtlebot3_autorace #17: error roslaunch turtlebot3_autorace_detect turtlebot3_autorace_detect_lane.launch

ji010, 2018-07-10:
I am learning turtlebot3 with ROS.

http://emanual.robotis.com/docs/en/platform/turtlebot3/autonomous_driving/#tutorials-44-settings-for-recognition
I am learning through here.

I have a problem while doing 13.2.5.1Tutorials: 5.1 Lane Detection.
roslaunch turtlebot3_autorace_detect turtlebot3_autorace_detect_lane.launch will cause an error.

The contents of the error are as follows.
[ERROR] [1531197022.305949]: bad callback: <bound method DetectLane.cbFindLane of <__ main __. DetectLane instance at 0x7ff7b58a9cf8 >>
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb (msg)
  File "/ home / cutop14 / catkin_ws / src / turtlebot3_autorace / turtlebot3_autorace_detect / nodes / detect_lane", line 170, in cbFindLane
    self.left_fit = np.array ([np.mean (self.mov_avg_left [:: - 1] [:, 0] [0: MOV_AVG_LENGTH]),
AttributeError: DetectLane instance has no attribute 'mov_avg_left'

This error will continue to occur until it stops running.

This is what I have done.

PC: roscore
SBS: roslaunch turtlebot3_autorace_camera turtlebot3_autorace_camera_pi.launch
          export AUTO_IN_CALIB = action
          roslaunch turtlebot3_autorace_camera turtlebot3_autorace_intrinsic_camera_calibration.launch
PC: export AUTO_EX_CALIB = action
        roslaunch turtlebot3_autorace_camera turtlebot3_autorace_extrinsic_camera_calibration.launch
        export AUTO_DT_CALIB = action
        roslaunch turtlebot3_autorace_detect turtlebot3_autorace_detect_lane.launch

If you know the solution to this problem, let me know.
Thanks

kijongGil, 2018-07-12:
Run the following command on SBC

' export AUTO_EX_CALIB = action
  roslaunch turtlebot3_autorace_camera turtlebot3_autorace_extrinsic_camera_calibration.launch'

ji010, 2018-07-12:
I got this error when I tried it on SBC.

Traceback (most recent call last):
  File / home / cubot14 / catkin_ws / src / turtlebot3_autorace / turtlebot3_autorace_camera / nodes / image_projection ", line 28, in <module>
    from turtlebot3_autorace_camera.cfg import ImageProjectionParamsConfig
ImportError: No module named turtlebot3_autorace_camera.cfg
Traceback (most recent call last):
  File / home / cubot14 / catkin_ws / src / turtlebot3_autorace / turtlebot3_autorace_camera / nodes / image_compensation ", line 28, in <module>
    from turtlebot3_autorace_camera.cfg import ImageCompensationParamsConfig
ImportError: No module named turtlebot3_autorace_camera.cfg
Traceback (most recent call last):
  File / home / cubot14 / catkin_ws / src / turtlebot3_autorace / turtlebot3_autorace_camera / nodes / image_compensation ", line 28, in <module>
    from turtlebot3_autorace_camera.cfg import ImageCompensationParamsConfig
ImportError: No module named turtlebot3_autorace_camera.cfg
camera / image_computer-1] process has died [pid 2296, exit code 1, cmd / home / cubot14 / catkin_ws / src / turtlebot3_autorace / turtlebot3_autorace_camera / nodes / image_compensation / camera / image_input: compressed: = / camera / image_rect_color / compressed / camera / image_output: = / camera / image_compensated / camera / image_output / compressed: = / camera / image_compensated / compressed__name: = image_compensation __log: = / home / cubot14 / 1577aab2-85a1-11e8-a31e-b827eb8c0976 / camera-image_compensation-1.log].
log file: /home/cubot14/.ros/log/1577aab2-85a1-11e8-a31e-b827eb8c0976/camera-image_compensation-1*.log
image_projection-2] process has died [pid 2297, exit code 1, cmd / home / cubot14 / catkin_ws / src / turtlebot3_autorace / turtlebot3_autorace_camera / nodes / image_projection / camera / image_input: compressed: = / camera / image_rect_color / compressed / camera / image_output: = / camera / image_projected / camera / image_output / compressed: = / camera / image_extrinsic_calib / compressed__name: = image_projection __log: = / home / cubot14 / .ros / log / 1577aab2-85a1-11e8-a31e-b827eb8c0976 / camera-image_projection-2.log].
log file: /home/cubot14/.ros/log/1577aab2-85a1-11e8-a31e-b827eb8c0976/camera-image_projection-2*.log
image_computation / camera / image_computation_projection-3] process has died [pid 2298, exit code 1, cmd / home / cubot14 / catkin_ws / src / turtlebot3_autorace / turtlebot3_autorace_camera / nodes / image_compensation / compressed: = / camera / image_projected / compressed / camera / image_output: = / camera / image_projected_compensated / camera / image_output / compressed: = / camera / image_projected_compensated / compressed__name: = image_compensation_projection __log: 1577aab2-85a1-11e8-a31e-b827eb8c0976 / camera-image_compensation_projection-3.log].
log file: /home/cubot14/.ros/log/1577aab2-85a1-11e8-a31e-b827eb8c0976/camera-image_compensation_projection-3*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor ...
... shutting down processing monitor complete
done

What should I do?

kijongGil, 2018-07-19:
I'm sorry for the late reply.
This problem is you can't import cfg files. The simplest way to solve this problem is using 'chmod' command. 
type follow command In any path where cfg files are located,
`$chmod +x cfg.file`

Thanks,
Gilbert

ji010, 2018-07-20:
Hi, I solved this problem.
It was difficult because I could not solve the error, but I really appreciate your help.
I really appreciate you.

kijongGil, 2018-07-24:
I will close this issue. You can reopen this issue when you have a issue.
Thanks.

closed 2018-07-24 by kijongGil

RIYAN-ROS, 2022-09-08:
$ colcon build --symlink-install

Starting >>> turtlebot3_msgs
Starting >>> dynamixel_sdk                     
Starting >>> hls_lfcd_lds_driver
Starting >>> turtlebot3_description
Finished <<< turtlebot3_description [3.08s]                               
Starting >>> dynamixel_sdk_custom_interfaces                              
Finished <<< dynamixel_sdk [3.26s]                                        
Starting >>> turtlebot3_cartographer
Finished <<< turtlebot3_cartographer [1.00s]                              
Starting >>> turtlebot3_gazebo
Finished <<< turtlebot3_msgs [8.01s]                                       
Starting >>> turtlebot3_node
Finished <<< dynamixel_sdk_custom_interfaces [5.61s]                          
Starting >>> turtlebot3_navigation2
Finished <<< turtlebot3_navigation2 [0.92s]                                   
Starting >>> turtlebot3_teleop
Finished <<< turtlebot3_gazebo [5.44s]                                        
Starting >>> turtlebot3_example
Finished <<< turtlebot3_teleop [14.4s]                                         
Starting >>> turtlebot3_fake_node
Finished <<< turtlebot3_example [14.4s]                                        
Starting >>> ld08_driver
Finished <<< ld08_driver [6.10s]                                               
Starting >>> dynamixel_sdk_examples
--- stderr: turtlebot3_node             
/home/riyan/turtlebot3_ws/src/turtlebot3/turtlebot3/turtlebot3_node/src/odometry.cpp: In constructor ‘robotis::turtlebot3::Odometry::Odometry(std::shared_ptr<rclcpp::Node>&, double, double)’:
/home/riyan/turtlebot3_ws/src/turtlebot3/turtlebot3/turtlebot3_node/src/odometry.cpp:39:58: error: no matching function for call to ‘rclcpp::Node::declare_parameter<std::string>(const char [18])’
   39 |   nh_->declare_parameter<std::string>("odometry.frame_id");
      |                                                          ^
In file included from /opt/ros/foxy/include/rclcpp/node.hpp:1224,
                 from /opt/ros/foxy/include/rclcpp/executors/single_threaded_executor.hpp:28,
                 from /opt/ros/foxy/include/rclcpp/executors.hpp:22,
                 from /opt/ros/foxy/include/rclcpp/rclcpp.hpp:146,
                 from /opt/ros/foxy/include/message_filters/subscriber.h:38,
                 from /home/riyan/turtlebot3_ws/src/turtlebot3/turtlebot3/turtlebot3_node/include/turtlebot3_node/odometry.hpp:25,
                 from /home/riyan/turtlebot3_ws/src/turtlebot3/turtlebot3/turtlebot3_node/src/odometry.cpp:17:
/opt/ros/foxy/include/rclcpp/node_impl.hpp:157:1: note: candidate: ‘auto rclcpp::Node::declare_parameter(const string&, const ParameterT&, const ParameterDescriptor&, bool) [with ParameterT = std::__cxx11::basic_string<char>; std::string = std::__cxx11::basic_string<char>; rcl_interfaces::msg::ParameterDescriptor = rcl_interfaces::msg::ParameterDescriptor_<std::allocator<void> >]’
  157 | Node::declare_parameter(
      | ^~~~
/opt/ros/foxy/include/rclcpp/node_impl.hpp:157:1: note:   candidate expects 4 arguments, 1 provided
/home/riyan/turtlebot3_ws/src/turtlebot3/turtlebot3/turtlebot3_node/src/odometry.cpp:40:64: error: no matching function for call to ‘rclcpp::Node::declare_parameter<std::string>(const char [24])’
   40 |   nh_->declare_parameter<std::string>("odometry.child_frame_id");
      |                                                                ^
In file included from /opt/ros/foxy/include/rclcpp/node.hpp:1224,
                 from /opt/ros/foxy/include/rclcpp/executors/single_threaded_executor.hpp:28,
                 from /opt/ros/foxy/include/rclcpp/executors.hpp:22,
                 from /opt/ros/foxy/include/rclcpp/rclcpp.hpp:146,
                 from /opt/ros/foxy/include/message_filters/subscriber.h:38,
                 from /home/riyan/turtlebot3_ws/src/turtlebot3/turtlebot3/turtlebot3_node/include/turtlebot3_node/odometry.hpp:25,
                 from /home/riyan/turtlebot3_ws/src/turtlebot3/turtlebot3/turtlebot3_node/src/odometry.cpp:17:
/opt/ros/foxy/include/rclcpp/node_impl.hpp:157:1: note: candidate: ‘auto rclcpp::Node::declare_parameter(const string&, const ParameterT&, const ParameterDescriptor&, bool) [with ParameterT = std::__cxx11::basic_string<char>; std::string = std::__cxx11::basic_string<char>; rcl_interfaces::msg::ParameterDescriptor = rcl_interfaces::msg::ParameterDescriptor_<std::allocator<void> >]’
  157 | Node::declare_parameter(
      | ^~~~
/opt/ros/foxy/include/rclcpp/node_impl.hpp:157:1: note:   candidate expects 4 arguments, 1 provided
/home/riyan/turtlebot3_ws/src/turtlebot3/turtlebot3/turtlebot3_node/src/odometry.cpp:42:50: error: no matching function for call to ‘rclcpp::Node::declare_parameter<bool>(const char [17])’
   42 |   nh_->declare_parameter<bool>("odometry.use_imu");
      |                                                  ^
In file included from /opt/ros/foxy/include/rclcpp/node.hpp:1224,
                 from /opt/ros/foxy/include/rclcpp/executors/single_threaded_executor.hpp:28,
                 from /opt/ros/foxy/include/rclcpp/executors.hpp:22,
                 from /opt/ros/foxy/include/rclcpp/rclcpp.hpp:146,
                 from /opt/ros/foxy/include/message_filters/subscriber.h:38,
                 from /home/riyan/turtlebot3_ws/src/turtlebot3/turtlebot3/turtlebot3_node/include/turtlebot3_node/odometry.hpp:25,
                 from /home/riyan/turtlebot3_ws/src/turtlebot3/turtlebot3/turtlebot3_node/src/odometry.cpp:17:
[... cut ...]
Failed   <<< turtlebot3_node [24.8s, exited with code 2]
Aborted  <<< turtlebot3_fake_node [14.1s]          
Aborted  <<< hls_lfcd_lds_driver [45.5s]           
Aborted  <<< dynamixel_sdk_examples [20.9s]                               
                                   
Summary: 10 packages finished [58.9s]
  1 package failed: turtlebot3_node
  3 packages aborted: dynamixel_sdk_examples hls_lfcd_lds_driver turtlebot3_fake_node
  3 packages had stderr output: hls_lfcd_lds_driver turtlebot3_fake_node turtlebot3_node
  3 packages not processed

trisha3290, 2024-04-07:
> Hi, I solved this problem. It was difficult because I could not solve the error, but I really appreciate your help. I really appreciate you.

Hello @Jeongin-park can you please tell how did you solve it?