ROBOTIS-GIT/turtlebot3_autorace #18: error turtlebnot3_autorace extrinsic camera calibration

seokyeongbaek, 2018-07-10:
hi 
An error occurred while following this e-manual. 

http://emanual.robotis.com/docs/en/platform/turtlebot3/autonomous_driving/#tutorials-43-extrinsic-camera-calibration

13.2.4.3 Tutorials: 4.3. Extrinsic Camera Caliaration
1. [PC]$ roscore
2. [SBC]$ roslaunch turtlebot3_autorace_camera turtlebot3_autorace_camera_pi.launch
after 
3.[SBC] $ export AUTO_IN_CALIB=action
   [SBC]$ roslaunch turtlebot3_autorace_camera turtlebot3_autorace_intrinsic_camera_calibration.launch
 there was an error here 

Error content

[ERROR] [1531198964.695398547]: Skipped loading plugin with error: XML Document '/opt/ros/kinetic/share/gmapping/nodelet_plugins.xml' has no Root Element. This likely means the XML is malformed or missing..

kijongGil, 2018-07-10:
Hi @seokyeongbaek :)
What is your ROS version? indigo or kinetic?

seokyeongbaek, 2018-07-10:
My ROS version is kinetic

kijongGil, 2018-07-10:
Maybe follow link will help you.
https://answers.ros.org/question/277049/gazebo-hangs-loading-turtlebot-world-skipping-xml-document-gmapping/

If you can not solve the problem, I recommend reinstalling ROS.

Thanks,
Gilbert

seokyeongbaek, 2018-07-11:
Thanks for your quick reply.!!!
I solved the problem. Thanks to you !!!!!!!!

closed 2018-07-11 by kijongGil