ROBOTIS-GIT/turtlebot3_autorace #20: 오류 roslaunch turtlebot3_autorace_detect turtlebot3_autorace_detect_sign.launch

ji010, 2018-07-20:
Hi, this problem has been solved by someone else but it has not been solved.
I want to be helped by this problem.

I'm getting an error in roslaunch turtlebot3_autorace_detect turtlebot3_autorace_detect_sign.launch.

[pc] roscore
[SBC] roslaunch turtlebot3_autorace_camera turtlebot3_autorace_camera_pi.launch
          export AUTO_IN_CALIB = action
          roslaunch turtlebot3_autorace_camera turtlebot3_autorace_intrinsic_camera_calibration.launch
          export AUTO_EX_CALIB = action
          roslaunch turtlebot3_autorace_camera turtlebot3_autorace_extrinsic_camera_calibration.launch
This is what I did before.
My turtle model is a burger.

If you look at the terminal after running launch, it is this.
process [detect_sign-1]: started with pid [10917]
OpenCV Error: Bad argument (image is empty or has incorrect depth (! = CV_8U)) in detectAndCompute, file /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/opencv_contrib/xfeatures2d/src/sift.cpp, line 1116
Traceback (most recent call last):
  File / home / cutop14 / catkin_ws / src / turtlebot3_autorace / turtlebot3_autorace_detect / nodes / detect_sign ", line 257, in <module>
    node = DetectSign ()
  File "/ home / cutop14 / catkin_ws / src / turtlebot3_autorace / turtlebot3_autorace_detect / nodes / detect_sign", line 34, in __init__
    self.fnPreproc ()
  File "/ home / cutop14 / catkin_ws / src / turtlebot3_autorace / turtlebot3_autorace_detect / nodes / detect_sign", line 73, in fnPreproc
    self.kp2, self.des2 = self.sift.detectAndCompute (self.img2, None)
cv2.error: /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/opencv_contrib/xfeatures2d/src/sift.cpp:1116: error: (-5) image is empty or has incorrect depth (! = CV_8U) in function detectAndCompute

detect_sign-1] process has died [pid 10917, exit code 1, cmd / home / cutop14 / catkin_ws / src / turtlebot3_autorace / turtlebot3_autorace_detect / nodes / detect_sign / detect / image_input: = / camera / image_compensated / detect / image_input / compressed: = detect / image_traffic_sign / detect / image_traffic_sign / detect / image_output / compressed: = / detect / image_traffic_sign / compressed__name: = detect_sign __log: = / home / cutop14 / .ros / log / 75997fba- 8bbe-11e8-87b6-b827eb8c0976 / detect_sign-1.log].
log file: /home/cutop14/.ros/log/75997fba-8bbe-11e8-87b6-b827eb8c0976/detect_sign-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor ...
... shutting down processing monitor complete
done

I took an image and put it in the right path.
(catkin_ws / src / turtlebot3_autorace / turtlebot3_autorace_detect / file / detect_sign)

I do not know what the problem is.
If you know the answer to this question, please help me.
Thank you.

kijongGil, 2018-07-20:
Hi @ji010 :)
The intrinsic and extrinsic calibration launch must be executed in Remote PC, not turtlebot3.
Please follow the manual carefully.
http://emanual.robotis.com/docs/en/platform/turtlebot3/autonomous_driving/#tutorials-41-camera-imaging-calibration

Thanks,
Gilbert

ji010, 2018-07-20:
Oh, I'm so sorry.
I wrote it wrong.
I ran the extrinsic calibration launch on my PC.
I did it on my PC, but I got an error.

Thanks

kijongGil, 2018-07-24:
I'm sorry to lately reply.
Did you calibrate detected lane?
Could you see the withe and yellow masks in rqt_image_view?
Because I think your problem is not match topic name between gazebo mode and real mode.

routiful, 2018-08-02:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2018-08-02 by routiful