ROBOTIS-GIT/turtlebot3_autorace #21: Bad callback errors occurred while detecting the lane, traffic sign, traffic light, etc.

wsung1, 2018-07-23:
Hi, ROBOTIS!
I'd thank you guys for such a great challenge you made!
Now I'm working on turtlebot3_autorace packages with following their tutorials (http://emanual.robotis.com/docs/en/platform/turtlebot3/autonomous_driving).
Nothing went wrong up to Tutorials: 4. Calibration.
However, from Tutorials: 5. Missions I've bumped into bad callback errors that look alike as follows.

5.1. Lane Detection
[image]

5.2. Traffic Sign
[image]

5.3. Traffic Light
[image]

5.4. Parking Lot
[image]

Do they mean that the callback function is not called because the message doesn't arrive at?
Please advise me how to approach this problem.
Thank you in advance!

kijongGil, 2018-07-24:
Hi @wsung1 :)
The problem is that the variable declaration position is incorrect. We will update this solve.
But if you want to solve right now, move 'self.counter = 1' on line 59 to line 38 for example in detect_lane. 'self.cvBridge = CvBridge()' is same. 
This issue is the part you found, so if you do a Pull Request we will get your PR. Do you want a PR? Or are we going to fix it? :)

Thanks,
Gilbert.

wsung1, 2018-07-24:
안녕하세요, 길버트님,
저는 조선대학교 기계시스템미래자동차공학부 성우석 교수입니다.
코드 수정 관련해서 전화통화 잠시 가능할지요?
감사합니다 :)

Jieul, 2018-07-24:
Hello, Gilbert.
I'm trying to your answer but I have a problem. In detect_lane.py, 
line 59 is 'self.sub_image_original = rospy.Subscriber('/detect/image_input', Image, self.cbFindLane, queue_size = 1)' 
and line 38 is 'self.lightness_white_h = rospy.get_param("~detect/lane/white/lightness_h", 255)'.
'self.counter = 1' on line 88. Where should I move 'self.counter = 1'?
Or did you mention another node?

wsung1, 2018-07-24:
Hi, @kijongGil,
I don't know why but the line number you mentioned mismatches with my detect_lane.py.
What about if you attach your detect_lane.py so that we could fix the code right away.
The regional competition is just around the corner, please help us move forward, @kijongGil :)

kijongGil, 2018-07-25:
I'm sorry for the confusion.
I said the line 38 and 59 are about detect_sign, In the case of detect_lane, self.counter = 1 is on line 88, move to 53 line.
It is an error that is referenced before the self.counter is define.

Jieul, 2018-07-25:
I changed detect_sign, detect_lane as your comment, but I'm still get the same error. Do you have any other solution to recommend?

Thanks, @kijongGil

Jieul, 2018-07-25:
The /catkin_ws/src/turtlebot3/turtlebot3_description/urdf/turtlebot3_burger_for_autorace.gazebo.xacro/ on [Remote PC], [SBC] was changed. Then I calibrated some camera matrix param(intrinsic, extrinsic) and each detection param(HSV) according to your comment. Each detection can be checked in rqt and it looks like it's working but the same error occurs.

Thanks for your reply @kijongGil !

kijongGil, 2018-08-07:
I'm sorry for late reply.
How about your problem?

closed 2018-08-13 by robotpilot