ROBOTIS-GIT/turtlebot3_autorace #22: Warn messages occurred while launching calibration

harry21105, 2018-07-24:
Hi everyone,
I got an warn error occurred while following everything on this e-manual.

http://emanual.robotis.com/docs/en/platform/turtlebot3/autonomous_driving/#tutorials-43-extrinsic-camera-calibration

Terminal code:
` $ roscore`
**Step1 TurtleBot SBC:**
` $ roslaunch turtlebot3_autorace_camera turtlebot3_autorace_camera_pi.launch`
**Step2 TurtleBot SBC:**
```
 $ export AUTO_IN_CALIB=action
 $ export GAZEBO_MODE=true ( I added it by myself )
 $ roslaunch turtlebot3_autorace_camera turtlebot3_autorace_intrinsic_camera_calibration.launch
```
**Step3 Remote PC:**
```
$ export AUTO_EX_CALIB=calibration
$ roslaunch turtlebot3_autorace_camera turtlebot3_autorace_extrinsic_camera_calibration.launch
```

After finish these steps, terminal in step2 got yellow warn messages... 
**Turtlebot SBC:**
```
[ WARN] [1532346786.215000834]: [image_transport] Topics '/camera/image_color' and '/camera/camera_info' do not appear to be synchronized. In the last 10s:
	Image messages received:      0
	CameraInfo messages received: 296
	Synchronized pairs:           0

```
Then the last step
` $ rqt`
Showed a pure white window without any information or image on it .

I also tried _Tutorials: 5.1 Lane Detection_ , but my turtlebot did not move at all. 
It showed the same yellow messages 

If there is any further information I need to show, please tell me. 
I would be as fast as I can.

kijongGil, 2018-07-24:
Hi @harry21105 :)
'export GAZEBO_MODE=true' means that you run autorace in gazebo.
But, I think you run autorace with actual turtlebot3 in real.
So, you don't enter this command or enter follow command.
`export GAZEBO_MODE=false` 

Thanks,
Gilbert.

harry21105, 2018-07-24:
Hi ;)
Thank you the problem is solved!!

closed 2018-07-24 by harry21105