ROBOTIS-GIT/turtlebot3_autorace #23: gazebo traffic lights cycle

yiorgosk, 2018-07-24:
Hello and congratulations for your great work. Based on your code i was able to spawn a second traffic light in gazebo and through some configurations i was able to make turtlebot stop at this traffic light and then continue with the parking and tunnel modes. The next step i want to take is to make turtlebot pass through the traffic lights while they are still green with a constant speed. I can configure the speed at which the robot moves in the lane but whatever the speed or the delay i put to the node core_node_mission the yellow light always spawns and the robot inevitably slows down. My question is how can i control the operation cycle of the traffic lights?﻿

kijongGil, 2018-07-24:
Hi @yiorgosk :)
You can modified the gazebo environment.
https://github.com/ROBOTIS-GIT/turtlebot3_autorace/blob/864131bbd59c037577a7535cc9b3a7924ff26d29/turtlebot3_autorace_core/nodes/core_node_mission#L115
This line is that you wanted.

Thanks,
Gilbert

yiorgosk, 2018-07-24:
I tried to add this if statement between lines 105 and 106 but the process of the green light becoming yellow never begins. I added the same if statement after the while loop in line 104 but the result is the same the chain reaction of self.traffic_state==1 never begins.

kijongGil, 2018-07-25:
When turtlebot3 move to x=0.5, y=1.75, the 'self.traffic_state' set to 1.
What did you add between lines 105 and 106?

yiorgosk, 2018-07-25:
while not rospy.is_shutdown():
            self.current_time = time.time()
            if self.traffic_state == 1:  # turn on yellow light
               if abs(self.current_time - time.time()) > 2:
                rospy.wait_for_service('gazebo/spawn_sdf_model')
                spawn_model_prox = rospy.ServiceProxy('gazebo/spawn_sdf_model', SpawnModel)
                spawn_model_prox('traffic_light_yellow', self.yellow_light_model, "robotos_name_space", self.initial_pose, "world")
                del_model_prox = rospy.ServiceProxy('gazebo/delete_model', DeleteModel)
                del_model_prox('traffic_light_green')
                self.traffic_state = 2
                self.current_time = time.time()

kijongGil, 2018-07-26:
How are you doing? I hope everything is going great.

yiorgosk, 2018-07-26:
Hello I am fine and you? In the lines 105 and 106 I added the code you
added above, the if statement in order to delay the yellow light from
spawning,  but the whole procedure never begins, the yellow light never
spawns and the robot sees the green light and continues all the way to the
end of the track. I also added before the if statement
self.current_time=time.time ()

kijongGil, 2018-08-07:
I'm sorry for late reply.
When turtlebot3 is approaching the traffic light, traffic light is working.
https://github.com/ROBOTIS-GIT/turtlebot3_autorace/blob/864131bbd59c037577a7535cc9b3a7924ff26d29/turtlebot3_autorace_core/nodes/core_node_mission#L45
So, reset `self.traffic_state = 1` at regular intervals. 
Thanks.

closed 2018-08-13 by robotpilot