ROBOTIS-GIT/turtlebot3_autorace #24: Problem in Tutorials: 4.1. Camera Imaging Calibration

wsung1, 2018-07-27:
Hi, ROBOTIS!
I run into a problem in walking through Tutorials: 4.1. Camera Imaging Calibration.
As advised, the parameter values were adjusted and then overwritten to the turtlebot3_autorace_camera/calibration/camera_calibration/camera.yaml.
From next launching, the adjusted values were shown in the terminal but they didn't work on rqt.
Thus, I've been redoing
$ rosrun rqt_reconfigure rqt_reconfigure
everytime I run rqt.
Otherwise, the rqt displays camera imaging with the default parameter values.
Do you know the reason why the values recorded in the yaml file are not reflected to rqt?
I'd appreciate any of ideas you'd share :)

robotpilot, 2018-08-07:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2018-08-07 by robotpilot

EndlessLoops, 2019-02-21:
hi，  @kijongGil 
I got the same problem in Camera Imaging Calibration and  Extrinsic Camera Ccalibration.I found it always use the defalut parameter values in rqt and can't load the parameter values which has been adjusted . For example ，I have been modified the projection.yaml，but it can't be automatically load in next launching.
I'd appreciate any of ideas you'd share :)