ROBOTIS-GIT/turtlebot3_autorace #29: unable to roslaunch intrinsic_camera_calibration.launch in turtlebot sbc

Ayensyarah, 2019-03-17:
roslaunch turtlebot3_autorace_camera turtlebot3_autorace_intrinsic_camera_calibration.launch
... logging to /home/pi/.ros/log/8e9ca8f6-486b-11e9-9a22-28b2bd9bc8f5/roslaunch-raspberrypi-1133.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.1.150:37315/

SUMMARY
========

PARAMETERS
 * /camera/image_proc/queue_size: 20
 * /rosdistro: kinetic
 * /rosversion: 1.12.13

NODES
  /camera/
    image_proc (image_proc/image_proc)
  /
    republish (image_transport/republish)

ROS_MASTER_URI=http://192.168.1.203:11311

process[republish-1]: started with pid [1142]
ERROR: cannot launch node of type [image_proc/image_proc]: can't locate node [image_proc] in package [image_proc]

RyanJaehyunShim, 2019-03-18:
@Ayensyarah 

The manual you are looking at is for Ubuntu Mate (I guess you might have installed Raspbian). The below URL is how to install Ubuntu Mate you would want to refer to. Reinstall the OS and follow the manual again, then it should work.

Let me know if you have any other questions,
Ryan

Ayensyarah, 2019-03-18:
@RyanJaehyunShim 

the manual tell us warning if we use raspberry pi 3b+  cannot responded to the ubuntu mate. should we proceed with the installation this OS ?

RyanJaehyunShim, 2019-03-18:
@Ayensyarah 

Here is the link that you would like to refer to. 
(https://ubuntu-mate.community/t/ubuntu-mate-for-raspberry-pi-3-model-b/15867/8)

Install Ubuntu Mate as mentioned on MaxVMH April 18.

Regards,
Ryan

Ayensyarah, 2019-03-20:
@RyanJaehyunShim 

tq so much!! finally i made it. do you know how to bypass screen login in ubuntu mate?  
btw thanks for your help

RyanJaehyunShim, 2019-03-21:
@Ayensyarah 

Morning. I do not fully understand your question, so you might want to send screenshots of your screen to help me understand the issue better. If you are having an issue with logging in, you will need to reinstall the OS to reset it. If you are asking how to change the login setup, I have no idea either but googling with some keywords like by-pass, login screen, etc. might give you some clues.

Have a nice day, 
Ryan

Ayensyarah, 2019-03-21:
@RyanJaehyunShim 

my camera extrinsic freeze when i run camera mood action. why?
i try the Tutorials: 4.4. Settings for Recognition. my intrinsic camera have no problem anymore. can you

Ayensyarah, 2019-03-21:
@RyanJaehyunShim 

well tq for that suggestion of by-pass...

RyanJaehyunShim, 2019-03-21:
@Ayensyarah 

Please open another issue if you have any other questions about TB3 and I will close this issue.

Regards, 
Ryan

closed 2019-03-21 by RyanJaehyunShim