ROBOTIS-GIT/turtlebot3_autorace #30: Error when roslaunch turtlebot3_autorace_camera turtlebot3_autorace_extrinsic_camera_calibration.launch

Ayensyarah, 2019-03-21:
i try Tutorials: 4.4. Settings for Recognition from http://emanual.robotis.com/docs/en/platform/turtlebot3/autonomous_driving/#turtlebot3-autorace-online-competition and got some error when change export AUTO_EX_CALIB=calibration to export AUTO_EX_CALIB=action. 

my /camera/image_extrinstic_calib/compressed freeze in rqt. 

i also got warning massage

[WARN] [1553134601.618210]: Could not process inbound connection: topic types do not match: [sensor_msgs/Image] vs. [sensor_msgs/CompressedImage]{'topic': '/camera/image_projected/compressed', 'tcp_nodelay': '0', 'md5sum': '060021388200f6f0f447d0fcd9c64743', 'type': 'sensor_msgs/Image', 'callerid': '/rqt_gui_cpp_node_10389'}

RyanJaehyunShim, 2019-04-09:
@Ayensyarah 

Please try it with export AUTO_EX_CALIB=calibration in Remote PC. It should be the same as tutorial 4.2 and 4.3.

Regards,
Ryan

ROBOTIS-Kayman, 2019-05-09:
This issue has been closed as there weren't recent activities. Please feel free to reopen this thread if there's any opinion to throw. Thanks.

closed 2019-05-09 by ROBOTIS-Kayman