ROBOTIS-GIT/turtlebot3_autorace #31: Unable to Parking

Ayensyarah, 2019-04-01:
unable to parking even the parking sign already detected.

can you help me?

RyanJaehyunShim, 2019-04-01:
@Ayensyarah 

Can you give us more details about your questions (i.e., which tutorial or error code)

Regards,
Ryan

Ayensyarah, 2019-04-01:
@RyanJaehyunShim 

theres no error when i roslaunch autorace_core and sent data rostopic pub -1 /core/decided_mode std_msgs/UInt8 "data: 2". its follow the line.

then it detect parking sign but not change to parking mode. still following the line

RyanJaehyunShim, 2019-04-01:
@Ayensyarah 

There is a misunderstanding about the mechanism. Once the robot detects a parking sign, it will start to count the number of dots on the line. When the robot finds the dotted line you can see on the racing map (by counting the number of dots), it will move to the parking lot.

Regards,
Ryan

Ayensyarah, 2019-04-01:
@RyanJaehyunShim 

the i should change the dotted value in which file?
detect_parking.py right?

RyanJaehyunShim, 2019-04-01:
@Ayensyarah 

You need the map.

Ayensyarah, 2019-04-01:
@RyanJaehyunShim 

how can i get the mapping? i mean how can i mapping the map.

RyanJaehyunShim, 2019-04-01:
@Ayensyarah 

Refer to 13.6.1 and 13.6.2 if you want to test it on a simulator. If you need a physical map, refer to the following link (https://github.com/ROBOTIS-GIT/autorace_track).

arixrobotics, 2019-04-04:
Is this where it counts and check the dotted line? https://github.com/ROBOTIS-GIT/turtlebot3_autorace/blob/master/turtlebot3_autorace_detect/nodes/detect_parking#L220 

If yes, then it triggers even if it sees only 1 dot? Or it counts until a few dots, then start to go in?

RyanJaehyunShim, 2019-04-09:
@arixrobotics 

Yes, the latter is correct. You can modify the code and try it again, then you will know it.

Regards,
Ryan

ROBOTIS-Kayman, 2019-05-09:
This issue has been closed as there weren't recent activities. Please feel free to reopen this thread if there's any opinion to throw. Thanks.

closed 2019-05-09 by ROBOTIS-Kayman