ROBOTIS-GIT/turtlebot3_autorace #32: No module named turtlebot3_autorace_camera.cfg #26

amakil76, 2019-04-22:
I try to follow your answer in above question
sudo chmod +x turtlebot3_autorace_camera.cfg
and get
chmod: cannot access 'turtlebot3_autorace_camera.cfg': No such file or directory

I tried changing to diffrent directory in the turtlebot3_autorace and also try to search the turtlebot3_autorace_camera.cfg but no success.

Reinstalling roscore also didnt work for me

RyanJaehyunShim, 2019-04-22:
@amakil76 

Let us know which tutorial (e.g., tutorial number) you are talking about. 

Regards,
Ryan

amakil76, 2019-04-23:
It wa the turlebot3 autorace
13.2.4.3 tutorial 4.3 extrinsic camera calibration step 4.

RyanJaehyunShim, 2019-04-24:
@amakil76 

I do not find any CFG file in the tutorial. Would you follow the tutorial written on the e-manual and let us know if you are stuck?

Regards,
Ryan

amakil76, 2019-04-28:
I reinstall everything linux ros turtlebot. Now everything works.

closed 2019-04-28 by amakil76