ROBOTIS-GIT/turtlebot3_autorace #33: Cannot install autorace dependent package in turtlebot.

amakil76, 2019-04-29:
I am using raspian in the turtlebot. When I follow the autorace tutorial 13.2.3 Tutorial 3
The installation fail because of no package found.
And when I run tutorial 4.3  step 3 on the turtlebot an error stated the image/proc node is not available.

RyanJaehyunShim, 2019-04-29:
@amakil76 

Install Ubuntu MATE for the tutorial. 
As for the second question, send screenshots of your error code.

amakil76, 2019-04-30:
Problem solve with ubuntu mate but
Now when running mission , the robot wont move if AUTO_EX_CALIB is set to action. The detect/image_lane only show white screen and 

[ERROR] [1556585371.836523]: bad callback: <bound method DetectLane.cbFindLane of <__main__.DetectLane instance at 0x7fe2ecf907e8>>
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/home/amr/catkin_ws/src/turtlebot3_autorace/turtlebot3_autorace_detect/nodes/detect_lane", line 172, in cbFindLane
    self.right_fit = np.array([np.mean(self.mov_avg_right[::1][:, 0][0:MOV_AVG_LENGTH]),
AttributeError: DetectLane instance has no attribute 'mov_avg_right'

Error is shown after autorace_core is launch

RyanJaehyunShim, 2019-04-30:
@amakil76 

Please refer to the link below and let us know if it doesn't work. 
(https://github.com/ROBOTIS-GIT/turtlebot3_autorace/issues/17)

If it doesn't work, add a reply with more details (e.g., screenshots of Terminal) about what you have done. 

Regards,
Ryan

amakil76, 2019-04-30:
[image]
[image]

I follow exactly from #17 

I still could not find the cfg file

amakil76, 2019-04-30:
Could not find the folder containing turtlebot3_autorace_camera.cfg file

RyanJaehyunShim, 2019-05-02:
@amakil76 

Type the followings in the terminal window and see if it works. 
$ cd ~/catkin_ws/src/turtlebot3_autorace/turtlebot3_autorace_camera/cfg
$ chmod +x *

If it does not work, you might have missed some instructions. Read the following tutorial 13.1~13.6 carefully, and follow it again.
(http://emanual.robotis.com/docs/en/platform/turtlebot3/autonomous_driving/#turtlebot3-autorace)

Regards,
Ryan

RyanJaehyunShim, 2019-05-16:
This issue has been closed as there weren't recent activities. Please feel free to reopen this thread if you have any other questions. Thanks.

closed 2019-05-16 by RyanJaehyunShim