ROBOTIS-GIT/turtlebot3_autorace #34: How is the /detect/lane value calculated?

joshsale, 2019-05-15:
How does the /detect/lane topic value get calculated? It seems to be the only input into the control_lane node, and I'm unsure of how it is calculated. The value is approximately 500 at the center of the lane, but is sensitive to any change in the turtlebot's position left or right. The value also seems sensitive to if the turtlebot turns. The turning part has been giving me an issue, as the control tends to over-correct for the error from the center of the lane and quickly goes unstable. Adjusting the gain values seems to keep the turtlebot on track longer, but I'm hoping to gain some insight as to how the input value gets calculated.

RyanJaehyunShim, 2019-05-16:
@joshsale 

Adjust Kp and Kd values in the following file and see if it works, 
(/catkin_ws/src/turtlebot3_autorace/turtlebot3_autorace_control/nodes/control_lane)

twist.angular.z is the variable that you might be interested in. 

Regards,
Ryan

ROBOTIS-Kayman, 2019-07-19:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2019-07-19 by ROBOTIS-Kayman