ROBOTIS-GIT/turtlebot3_autorace #36: Unable to run Camera Imaging Calibration (Item 13.2.3.1 Tutorials: 4.1

red24dog, 2019-08-16:
After completing eManual Tutorials: 1 Requirements with current github source and compilation, and executing 4. Calibration Item 1. 2. 3. the following resulted in item 4. **Remote PC**:

turtlebot@herschel:~$ rosrun rqt_reconfigure rqt_reconfigure
Failed to load Python extension for LZ4 support. LZ4 compression will not be available.
Could not import "pyqt" bindings of qt_gui_cpp library - so C++ plugins will not be available:
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui_cpp/cpp_binding_helper.py", line 43, in <module>
    from . import libqt_gui_cpp_sip
ImportError: dynamic module does not define module export function (PyInit_libqt_gui_cpp_sip)

Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_reconfigure/node_selector_widget.py", line 275, in _selection_changed_slot
    self._selection_selected(index_current, rosnode_name_selected)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_reconfigure/node_selector_widget.py", line 225, in _selection_selected
    item_widget = item_child.get_dynreconf_widget()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_reconfigure/treenode_qstditem.py", line 154, in get_dynreconf_widget
    self._param_name_raw)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_reconfigure/dynreconf_client_widget.py", line 90, in __init__
    self.reconf.config_callback = self.config_callback
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/dynamic_reconfigure/client.py", line 276, in set_config_callback
    self._config_callback(self.config)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_reconfigure/dynreconf_client_widget.py", line 106, in config_callback
    rospy.logdebug('config_callback name={} v={}'.format(name, v))
NameError: name 'name' is not defined
Aborted (core dumped)
turtlebot@herschel:~$

rjshim, 2019-08-20:
@red24dog 

Please refer to the following links ([link1](https://github.com/RobotWebTools/rosbridge_suite/issues/384), [link2](
https://answers.ros.org/question/247389/rqt_graph-error-could-not-import-pyqt-bindings-of-qt_gui_cpp-library-so-c-plugins-will-not-be-available/)) It seems like the problem is on python. Remove all python related packages and reinstall it following [this instruction](http://emanual.robotis.com/docs/en/platform/turtlebot3/machine_learning/#anaconda)

If it still doesn't work, reinstall your OS and try it again and let us know the result.

Regards,
Ryan

red24dog, 2019-08-22:
Solved by your suggested action to remove python related packages. [not related directly to Turtlebot3] and ~$ rqt reconfigure reconfigure works. Problem was several months ago, I installed anaconda3 on my **Remote PC**, which was a bad idea. It installed its own Python 3.7.4 to the Home directory, added a script to the .bashrc which expropriated "python" to this python3. I uninstalled anaconda3 (Anaconda-Clean) , deleted the ~/anaconda3 folder and restored the .bashrc to the turtlebot3 eManual defined version. I did not need to reinstall the OS(Ubuntu 16.04) or python2.7.12. After this action, ~$ which python, python2, python3 displayed /usr/bin/python, /usr/bin/python2, python3 /usr/bin/python3, respectively. Lesson is to be careful installing any python packages in the **Remote PC** (and **SBC**) . Thank you Ryan

closed 2019-08-22 by red24dog