ROBOTIS-GIT/turtlebot3_autorace #37: Unable to run 13.2.5. Tutorials: 5. Missions

red24dog, 2019-08-16:
Refereing to my previous post #36 on this forum, "Unable to run Camera Imaging Calibration (Item 13.2.3.1 Tutorials: 4.1), regardless of the error I attempted to run the 5.1 Lane Deection, with the following transcript from the **Remote PC** (herschel), indication an issue with the cv2 code, PyCObject_Type? Please note that "roscore" executes without error, "rosnode list" shows /rosout.

turtlebot@herschel:~$ export AUTO_DT_CALIB=calibration
turtlebot@herschel:~$ roslaunch turtlebot3_autorace_detect turtlebot3_autorace_detect_lane.launch
... logging to /home/turtlebot/.ros/log/4b706308-c078-11e9-acd8-60d819ca036d/roslaunch-herschel-25942.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://herschel.local:44081/

SUMMARY
========

PARAMETERS
 * /detect_lane/detect/lane/white/hue_h: 25
 * /detect_lane/detect/lane/white/hue_l: 0
 * /detect_lane/detect/lane/white/lightness_h: 255
 * /detect_lane/detect/lane/white/lightness_l: 180
 * /detect_lane/detect/lane/white/saturation_h: 36
 * /detect_lane/detect/lane/white/saturation_l: 0
 * /detect_lane/detect/lane/yellow/hue_h: 41
 * /detect_lane/detect/lane/yellow/hue_l: 27
 * /detect_lane/detect/lane/yellow/lightness_h: 255
 * /detect_lane/detect/lane/yellow/lightness_l: 160
 * /detect_lane/detect/lane/yellow/saturation_h: 255
 * /detect_lane/detect/lane/yellow/saturation_l: 130
 * /detect_lane/is_detection_calibration_mode: True
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    detect_lane (turtlebot3_autorace_detect/detect_lane)

ROS_MASTER_URI=http://herschel.local:11311

process[detect_lane-1]: started with pid [25959]
Traceback (most recent call last):
  File "/home/turtlebot/catkin_ws/src/turtlebot3_autorace/turtlebot3_autorace_detect/nodes/detect_lane", line 24, in <module>
    import cv2
ImportError: /opt/ros/kinetic/lib/python2.7/dist-packages/cv2.so: undefined symbol: PyCObject_Type
[detect_lane-1] process has died [pid 25959, exit code 1, cmd /home/turtlebot/catkin_ws/src/turtlebot3_autorace/turtlebot3_autorace_detect/nodes/detect_lane /detect/image_input:=/camera/image_projected_compensated /detect/image_input/compressed:=/camera/image_projected_compensated/compressed /detect/image_output:=/detect/image_lane /detect/image_output/compressed:=/detect/image_lane/compressed /detect/image_output_sub1:=/detect/image_white_lane_marker /detect/image_output_sub1/compressed:=/detect/image_white_lane_marker/compressed /detect/image_output_sub2:=/detect/image_yellow_lane_marker /detect/image_output_sub2/compressed:=/detect/image_yellow_lane_marker/compressed __name:=detect_lane __log:=/home/turtlebot/.ros/log/4b706308-c078-11e9-acd8-60d819ca036d/detect_lane-1.log].
log file: /home/turtlebot/.ros/log/4b706308-c078-11e9-acd8-60d819ca036d/detect_lane-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
turtlebot@herschel:~/

rjshim, 2019-08-20:
@red24dog 

As far as what [this issue thread](https://github.com/facebookresearch/faiss/issues/609) says, it seems like the problem is occurring because of python. Reinstall python, try it again and let us know the result.

Regards,
Ryan

red24dog, 2019-08-20:
Thanks, and I ask for clarification: Reinstall which python, 2.7 or 3.6 or both?

Do you have a recommended url for instructions to to this?

Ross Lunan

red24dog, 2019-08-22:
Solved as described in my previous Comment posting #36. Removed anaconda3 and restore .bashrc.

closed 2019-08-22 by red24dog