ROBOTIS-GIT/turtlebot3_autorace #43: Errors when 13.2.6 Tutorials: 6. Run Autonomous Driving

red24dog, 2019-08-22:
Item 4. Turtlebot SBC, "GAZEBO_MODE is not set"
roslaunch turtlebot3_autorace_camera turtlebot3_autorace_intrinsic_camera_calibration.launch
... logging to /home/turtlebot/.ros/log/5efed220-c518-11e9-a75c-60d819ca036d/roslaunch-cassini-2036.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

Invalid <arg> tag: environment variable 'GAZEBO_MODE' is not set. 

Arg xml is <arg default="$(env GAZEBO_MODE)" doc="gazebo_mode type [true, false]" name="use_gazebo"/>
The traceback for the exception was written to the log file

Trying :$export GAZEBO_MADE=true or false results in warning:
The input topic '/camera/camera_info' is not yet advertised

turtlebot@cassini:~$ roslaunch turtlebot3_autorace_camera turtlebot3_autorace_intrinsic_camera_calibration.launch
... logging to /home/turtlebot/.ros/log/5efed220-c518-11e9-a75c-60d819ca036d/roslaunch-cassini-2090.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://cassini.local:35223/

SUMMARY
========

PARAMETERS
 * /camera/image_proc/queue_size: 20
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /camera/
    image_proc (image_proc/image_proc)
  /
    republish (image_transport/republish)

ROS_MASTER_URI=http://herschel.local:11311

process[republish-1]: started with pid [2101]
process[camera/image_proc-2]: started with pid [2102]
[ WARN] [1566504976.719596878]: The input topic '/camera/camera_info' is not yet advertised
^C[camera/image_proc-2] killing on exit
[republish-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
turtlebot@cassini:~$ 

On Remote PC, Running :$ rqt_image_view shows no camera image

rjshim, 2019-08-23:
@red24dog 

Did you get any error codes from item2, 3, or maybe you skipped either of them? Show me your terminal windows.

Regards,
Ryan

red24dog, 2019-08-23:
No errors from 13.2.6: Item 2: roscore or Item 3 roslaunch turtlebot3_bringup turtlebot3_robot.launch and LIDAR spins. I should mention that I can run the 7. Bringup  7.1 roscore on Remote PC and 7.2 roslaunch turtlebot3_bringup turtlebot3_robot.launch, start the raspicam_node on SBC and see the image with rqt_view_image on Remote PC. Where does the GAZEBO_MODE get set?

red24dog, 2019-08-23:
Following up on your Aug 22 Comment, I went back to the calibration start,  13.2.4 Tutorials: 4.  Calibration. 

13.2.4.1 Tutorials: 4.1 Camera Calibration Steps 1, 2, 3, 4, ran without errors.

13.2.4.2. Tutorials 4.2 Intrisic Camera Calibration.
Item 1. Remote PC
turtlebot@herschel:~$ roscore
... logging to /home/turtlebot/.ros/log/0c742082-c5f9-11e9-ad16-60d819ca036d/roslaunch-herschel-4413.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://herschel.local:38401/
ros_comm version 1.12.14

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES

auto-starting new master
process[master]: started with pid [4423]
ROS_MASTER_URI=http://herschel.local:11311/

setting /run_id to 0c742082-c5f9-11e9-ad16-60d819ca036d
process[rosout-1]: started with pid [4436]
started core service [/rosout]

Item2. SBC
turtlebot@cassini:~$ roslaunch turtlebot3_autorace_camera turtlebot3_autorace_camera_pi.launch
... logging to /home/turtlebot/.ros/log/0c742082-c5f9-11e9-ad16-60d819ca036d/roslaunch-cassini-2043.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://cassini.local:46451/

SUMMARY
========

PARAMETERS
 * /camera/camera/ISO: 889
 * /camera/camera/awb_mode: tungsten
 * /camera/camera/brightness: 59
 * /camera/camera/contrast: 50
 * /camera/camera/exposureCompensation: 0
 * /camera/camera/exposure_mode: antishake
 * /camera/camera/hFlip: False
 * /camera/camera/saturation: 0
 * /camera/camera/sharpness: 0
 * /camera/camera/shutterSpeed: 25000
 * /camera/camera/vFlip: False
 * /camera/camera/videoStabilisation: False
 * /camera/camera/zoom: 1.0
 * /camera/camera_frame_id: camera
 * /camera/camera_info_url: package://turtleb...
 * /camera/framerate: 30
 * /camera/height: 240
 * /camera/width: 320
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    camera (raspicam_node/raspicam_node)

ROS_MASTER_URI=http://herschel.local:11311

process[camera-1]: started with pid [2052]
[ INFO] [1566601553.427335543]: Loading CameraInfo from package://turtlebot3_autorace_camera/calibration/intrinsic_calibration/camerav2_320x240_30fps.yaml
[ INFO] [1566601553.864375405]: Camera component done

[ INFO] [1566601553.870767109]: Encoder component done

[ INFO] [1566601553.885333585]: camera calibration URL: package://turtlebot3_autorace_camera/calibration/intrinsic_calibration/camerav2_320x240_30fps.yaml
[ INFO] [1566601554.022193994]: Camera successfully calibrated
[ INFO] [1566601554.850267264]: Reconfigure Request: contrast 0, sharpness 0, brightness 50, saturation 0, ISO 400, exposureCompensation 0, videoStabilisation 0, vFlip 0, hFlip 0, zoom 1.00, exposure_mode auto, awb_mode auto
[ INFO] [1566601554.852776686]: Reconfigure done
[ INFO] [1566601555.143278818]: Starting video capture (320, 240, 80, 30)

[ INFO] [1566601555.144411628]: Video capture started

Item 3. Print Checkerboard-Done

Item 4. Remote PC

red24dog, 2019-08-23:
Item 4. Remote PC
turtlebot@cassini:~$ export AUTO_IN_CALIB=calibrate
turtlebot@cassini:~$ roslaunch turtlebot3_autorace_camera turtlebot3_autorace_intrinsic_camera_calibration.launch
... logging to /home/turtlebot/.ros/log/0c742082-c5f9-11e9-ad16-60d819ca036d/roslaunch-cassini-2250.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

Invalid <arg> tag: environment variable 'GAZEBO_MODE' is not set. 

Arg xml is <arg default="$(env GAZEBO_MODE)" doc="gazebo_mode type [true, false]" name="use_gazebo"/>
The traceback for the exception was written to the log file
turtlebot@cassini:~$ export GAZEBO_MODE=false
turtlebot@cassini:~$ roslaunch turtlebot3_autorace_camera turtlebot3_autorace_intrinsic_camera_calibration.launch
... logging to /home/turtlebot/.ros/log/0c742082-c5f9-11e9-ad16-60d819ca036d/roslaunch-cassini-2256.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://cassini.local:38459/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    republish (image_transport/republish)

ROS_MASTER_URI=http://herschel.local:11311

process[republish-1]: started with pid [2266]

*** HUNG HERE***

^C[republish-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

turtlebot@cassini:~$ rosnode list
/camera
/cameracalibrator
/rosout
turtlebot@cassini:~$ roswtf
Loaded plugin tf.tfwtf
No package or stack in context
================================================================================
Static checks summary:

No errors or warnings
================================================================================
Beginning tests of your ROS graph. These may take awhile...
analyzing graph...
... done analyzing graph
running graph rules...
ERROR: connection refused to [http://herschel.local:44945/]
... done running graph rules

Online checks summary:

Found 1 warning(s).
Warnings are things that may be just fine, but are sometimes at fault

WARNING The following node subscriptions are unconnected:
 * /cameracalibrator:
   * /camera/image
   * /left
   * /right

Found 3 error(s).

ERROR Could not contact the following nodes:
 * /cameracalibrator

ERROR The following nodes should be connected but aren't:
 * /cameracalibrator->/rosout (/rosout)

ERROR Errors connecting to the following services:
 * service [/cameracalibrator/set_logger_level] appears to be malfunctioning: Unable to communicate with service [/cameracalibrator/set_logger_level], address [rosrpc://herschel.local:38983]
 * service [/cameracalibrator/get_loggers] appears to be malfunctioning: Unable to communicate with service [/cameracalibrator/get_loggers], address [rosrpc://herschel.local:38983]

turtlebot@cassini:~$

rjshim, 2019-08-26:
@red24dog 

You will need to set GAZEBO_MODE to false or true depending on whether you are operating a robot on the Gazebo simulator environment or your actual physical robot. It seems like 13 Autonomous Driving section is outdated so I suggest you skip it if it is not necessary. 

Regards,
Ryan

red24dog, 2019-08-26:
Are you saying that all of Section 13 code and instructions , Autonomous Driving is outdated, which then means Autorace functions is not possible? Can ROBOTIS update the Autorace code? I have tried to run just the Section 13.2.6: Tutorial 6, Run Autonomous Driving and encountered error messages, printed in the start of this comment thread,

 [ WARN] [1566504976.719596878]: The input topic '/camera/camera_info' is not yet advertised
^C[camera/image_proc-2] killing on exit
[republish-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete

Can you suggest ROS diagnostic commands  that I could do to find the camera problem? The Turtlebot3 (including camera) works well with basic teleoperation and SLAM/Navigation.

rjshim, 2019-08-26:
@red24dog 

I deleted my previous answer as it was wrong. I tested the entire autonomous driving section and there were only a few mistakes in the explanation so it will work fine if you do it in the right way. 

I updated the eManual so please read it and try it again. The difference is you will have to run 13.2.4.3, 13.2.4.4, 13.2.6 on Remote PC, not on TurtleBot PC.

To run 13.2.6, you will need an actual robot and map (with lanes traffic signs). Instead, you can test the package with Gazebo simulator, which is elaborated in subsection 13.6

Let me know if it still doesn't work, 
Ryan