ROBOTIS-GIT/turtlebot3_autorace #46: Camera's parameters that have been set from last calibration are not working in the next launching

wisnumurti7, 2019-10-13:
Excuse me, I am here as I have a little problem about the turltebot. I am looking for a solution here. I hope you can help me.

In Turtlebot3 autorace calibration tutorial from 13. 2. 4. 1 throgh 13. 2. 5. 1, it is said that overwrite each values from the calibration on to the <file.yaml> then it will make the camera set its parameters as you set here from next launching. In my case, those parameters is not working for the next launch (after I close connection from my remote computer to sbc or I reboot the sbc). I had tried using the automatic way (using rosrun rqt_reconfigure rqt_reconfigure) or manual way (overwriting the .yaml file), but it still not working. Is there any step that I miss? Do I have to re-calibrate the camera each time I start the turlebot?

I am looking up for you answer. Thank you

rjshim, 2019-10-15:
@wisnumurti7 

Provide us with more details about the issue. Which lines in which file have you modified and what error codes did you see when you run camera_pi next? Also filling in the form below will help us figure out the problem.

Regards,
Ryan

-----------------------------
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?

    - [ ] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] etc (PLEASE, WRITE DOWN YOUR ROS VERSION HERE)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

4. Which OS you installed in SBC?

    - [ ] Raspbian
    - [ ] Ubuntu MATE 16.04 or later
    - [ ] Ubuntu Server 18.04 or later
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] Linux Mint 19.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

6. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Write down the commands you used in order

    - HERE

8. Copy and Paste your error message on terminal

    - HERE

9. Please, describe detailedly what difficulty you are in

    - HERE

rjshim, 2019-10-21:
As there have been no updates, I am closing this issue.

closed 2019-10-21 by rjshim