ROBOTIS-GIT/turtlebot3_autorace #47: Autorace with gazebo I can't run it

Kontanin, 2020-01-03:
I can't run it so this is my error .
...............................................................................................................................................................................................................Step1
ros@ubuntu:~$ roslaunch turtlebot3_gazebo turtlebot3_autorace.launch
... logging to /home/ros/.ros/log/e05aa70c-2df7-11ea-a833-000c29ec8ba4/roslaunch-ubuntu-2725.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.43.108:38403/

SUMMARY
========

PARAMETERS
 * /robot_description: <?xml version="1....
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /use_sim_time: True

NODES
  /
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)
    spawn_urdf (gazebo_ros/spawn_model)

auto-starting new master
process[master]: started with pid [2740]
ROS_MASTER_URI=http://192.168.43.108:11311

setting /run_id to e05aa70c-2df7-11ea-a833-000c29ec8ba4
process[rosout-1]: started with pid [2753]
started core service [/rosout]
process[gazebo-2]: started with pid [2756]
process[gazebo_gui-3]: started with pid [2758]
process[spawn_urdf-4]: started with pid [2766]
[ INFO] [1578035341.839805010]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1578035341.906702998]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1578035341.914791494]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1578035341.949213288]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
Warning [parser.cc:778] XML Element[static], child of element[model] not defined in SDF. Ignoring[static]. You may have an incorrect SDF file, or an sdformat version that doesn't support this element.
Warning [parser.cc:778] XML Element[static], child of element[model] not defined in SDF. Ignoring[static]. You may have an incorrect SDF file, or an sdformat version that doesn't support this element.
Warning [parser.cc:778] XML Element[static], child of element[model] not defined in SDF. Ignoring[static]. You may have an incorrect SDF file, or an sdformat version that doesn't support this element.
Warning [parser.cc:778] XML Element[static], child of element[model] not defined in SDF. Ignoring[static]. You may have an incorrect SDF file, or an sdformat version that doesn't support this element.
Unhandled exception in thread started by 
sys.excepthook is missing
lost sys.stderr
[spawn_urdf-4] process has finished cleanly
log file: /home/ros/.ros/log/e05aa70c-2df7-11ea-a833-000c29ec8ba4/spawn_urdf-4*.log
[ INFO] [1578035350.856309041, 116.346000000]: Camera Plugin: Using the 'robotNamespace' param: '/'
[ INFO] [1578035350.943977367, 116.346000000]: Camera Plugin (ns = /)  <tf_prefix_>, set to ""
[ INFO] [1578035353.987588053, 116.346000000]: Laser Plugin: Using the 'robotNamespace' param: '/'
[ INFO] [1578035353.987730341, 116.346000000]: Starting Laser Plugin (ns = /)
[ INFO] [1578035354.032418320, 116.346000000]: Laser Plugin (ns = /)  <tf_prefix_>, set to ""
[ INFO] [1578035354.116655246, 116.346000000]: Starting plugin DiffDrive(ns = //)
[ INFO] [1578035354.116984969, 116.346000000]: DiffDrive(ns = //): <rosDebugLevel> = na
[ INFO] [1578035354.119639550, 116.346000000]: DiffDrive(ns = //): <tf_prefix> = 
[ INFO] [1578035354.156448481, 116.346000000]: DiffDrive(ns = //): Advertise joint_states
[ INFO] [1578035354.158373099, 116.346000000]: DiffDrive(ns = //): Try to subscribe to cmd_vel
[ INFO] [1578035354.189618850, 116.346000000]: DiffDrive(ns = //): Subscribe to cmd_vel
[ INFO] [1578035354.191487272, 116.346000000]: DiffDrive(ns = //): Advertise odom on odom 
[ INFO] [1578035354.268649437, 116.346000000]: imu plugin missing <frameName>, defaults to <bodyName>
[ INFO] [1578035354.377701082, 116.367000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1578035354.416524744, 116.377000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1578035354.781335030, 116.426000000]: Physics dynamic reconfigure ready.
[ INFO] [1578035354.870043617, 116.427000000]: Physics dynamic reconfigure ready.
gzserver: /build/ogre-1.9-mqY1wq/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreRenderSystem.cpp:546: virtual void Ogre::RenderSystem::setDepthBufferFor(Ogre::RenderTarget*): Assertion `bAttached && "A new DepthBuffer for a RenderTarget was created, but after creation" "it says it's incompatible with that RT"' failed.

Step2

roslaunch turtlebot3_gazebo turtlebot3_autorace_mission.launch
[... 215 lines cut ...]
========

PARAMETERS
 * /camera/image_proc/queue_size: 20
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /camera/
    image_proc (image_proc/image_proc)
  /
    relay_camera_info (topic_tools/relay)
    republish (image_transport/republish)

ROS_MASTER_URI=http://192.168.43.108:11311

process[republish-1]: started with pid [3817]
process[relay_camera_info-2]: started with pid [3892]
process[camera/image_proc-3]: started with pid [3902]

Step5

ros@ubuntu:~$ rostopic pub -1 /core/decided_mode std_msgs/UInt8 "data: 2"
publishing and latching message for 3.0 seconds
ros@ubuntu:~$

ROBOTIS-Will, 2020-01-22:
@Kontanin 
Could you post the actual error message displayed on the terminal?
I couldn't find the error message in your posting.

ROBOTIS-Will, 2020-02-28:
This issue has been closed as there weren't recent activities. Please feel free to reopen this thread if necessary. Thanks.

closed 2020-02-28 by ROBOTIS-Will