ROBOTIS-GIT/turtlebot3_autorace #48: IOError: [Errno 2] No such file or directory: '/opt/ros/melodic/lib/turtlebot3_autorace_core/traffic_light_red/model.sdf'

aliozcan01, 2020-04-20:
Hello,

I want to practice the link but I get an error 
http://emanual.robotis.com/docs/en/platform/turtlebot3/autonomous_gazebo/

"roslaunch turtlebot3_gazebo turtlebot3_autorace.launch " command is working, not error but ,
"roslaunch turtlebot3_gazebo turtlebot3_autorace_mission.launch" command is not working, get error.

I would be happy if you help ,thanks

started roslaunch server http://ali:36213/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.5

NODES
  /
    core_node_mission (turtlebot3_autorace_core/core_node_mission)

ROS_MASTER_URI=http://localhost:11311

process[core_node_mission-1]: started with pid [9603]
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/turtlebot3_autorace_core/core_node_mission", line 163, in <module>
    main()
  File "/opt/ros/melodic/lib/turtlebot3_autorace_core/core_node_mission", line 158, in main
    controlmission = ControlMission()
  File "/opt/ros/melodic/lib/turtlebot3_autorace_core/core_node_mission", line 38, in __init__
    self.loadMissionModel()
  File "/opt/ros/melodic/lib/turtlebot3_autorace_core/core_node_mission", line 65, in loadMissionModel
    red_light_model = open(red_light_path,'r')
IOError: [Errno 2] No such file or directory: '/opt/ros/melodic/lib/turtlebot3_autorace_core/traffic_light_red/model.sdf'
[core_node_mission-1] process has died [pid 9603, exit code 1, cmd /opt/ros/melodic/lib/turtlebot3_autorace_core/core_node_mission __name:=core_node_mission __log:=/home/ali/.ros/log/0979ac9a-834e-11ea-a2d0-28c2dd84afd9/core_node_mission-1.log].
log file: /home/ali/.ros/log/0979ac9a-834e-11ea-a2d0-28c2dd84afd9/core_node_mission-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

ROBOTIS-Will, 2020-04-21:
@aliozcan01 
First of all, the simulation is written based on ROS 1 Kinetic so there could be some differences in Melodic.
It seems like you haven't properly installed the package.
Make sure that you have all the necessary packages under the ~/catkin_ws/src folder.
As of now, binary installation using `apt install ros-kinetic....` does not provide the latest source.
Please use `git clone` method to download the source code and build locally.
Thank you.

aliozcan01, 2020-04-21:
Thank you for your answer.  
I do not have a missing package inside ~/catkin_ws/src . 
But I think there is a problem with ros melodic files.
There are various  Turtlebot3 files in their directory:
/opt/ros/melodic/lib
/opt/ros/melodic/share
sudo apt-get install ros-melodic-turtlebot3-*  
I installed it with this command
How can I install differently as you mentioned ?
Thanks.
Note:
My only problem is Autorace in Turtlebot3 Simulations. Other simulation sections work smoothly, with minor exceptions.

ROBOTIS-Will, 2020-04-28:
@aliozcan01 
Hi, we are currently preparing for the binary source code update.
In order to download the most up to date source code, please use the instruction with `git clone` command written in the eManual.
Please also note that there are 2 different Autorace version (2020 and 2019).
[http://emanual.robotis.com/docs/en/platform/turtlebot3/autonomous_driving/#ros-1-autonomous-driving](http://emanual.robotis.com/docs/en/platform/turtlebot3/autonomous_driving/#ros-1-autonomous-driving)
Thank you.

aliozcan01, 2020-05-01:
problem solved, thanks a lot

closed 2020-05-01 by aliozcan01

MayurRabadiya, 2020-08-10:
Hello aliozcan01,

i get same error.
how did you solve that error?

aliozcan01, 2020-08-10:
Hello Mayur,

I solved the problem by following the steps below. I hope that will be useful.

sudo apt-get install ros-melodic-turtlebot3-*
cd ~/catkin_ws/src/
git clone https://github.com/ROBOTIS-GIT/turtlebot3_msgs.git
git clone https://github.com/ROBOTIS-GIT/turtlebot3.git
git clone https://github.com/ROBOTIS-GIT/turtlebot3_simulations
git clone https://github.com/ROBOTIS-GIT/turtlebot3_autorace
cd ~/catkin_ws && catkin_make

MayurRabadiya, 2020-08-11:
> Hello Mayur,
>
> I solved the problem by following the steps below. I hope that will be useful.
>
> sudo apt-get install ros-melodic-turtlebot3-*
> cd ~/catkin_ws/src/
> git clone [...]

thanks for your reply.
But it does not work.

sume-cn, 2021-01-05:
I can run , do not use
`sudo apt-get install ros-melodic-turtlebot3-*`

after git clone and catkin_make:
`cd catkin_ws
source devel/setup.bash`
then roslaunch those launch files

ROBOTIS-Will, 2021-01-05:
@MayurRabadiya  Which ROS distribution are you using? The current Autorace packages support ROS 1 Kinetic and Melodic.

MayurRabadiya, 2021-01-06:
i am use melodic