ROBOTIS-GIT/turtlebot3_autorace #49: image_color and camera_info are not synchronized

krizchong, 2020-08-13:
When I running this

 $ export AUTO_IN_CALIB=action
 $ roslaunch turtlebot3_autorace_camera turtlebot3_autorace_intrinsic_camera_calibration.launch

There are always pop out the warning as:

[ WARN] [1597283229.652176080]: [image_transport] Topics '/camera/image_color' and '/camera/camera_info' do not appear to be synchronized. In the last 10s:
	Image messages received:      62
	CameraInfo messages received: 299
	Synchronized pairs:           0

Sometime the pairs is not 0, then the image process still working, but sometime it always 0...

Why this happen? I try to use different kind of WiFi connection, like smartphone hot spot or  WiFi router, also the same.
I also already ntpdate on both SBC and Remote PC. How to solve this issue?

BENBEY-I, 2022-06-03:
I had the same problem, I solved it using a wired connection, at the beginning I tried the wifi connection (the smartphone hotspot and the pc hotspot), with the smartphone hotspot the robot and the PC connected perfectly but I had this problem and with the PC hotspot the robot could not connect, so I tried the wired connection and it worked.

selvavignesh-2206, 2024-09-16:
Is there any update on this issue? I am facing the similar issue with local network WIFI.

yun-goon, 2025-03-10:
Sorry for the late response. This issue was created a long time ago.

The latest version has been updated, so please try again with the latest version. If you still encounter any problems, feel free to open a new issue.

Thank you for your understanding!