ROBOTIS-GIT/turtlebot3_autorace #51: [detect_parking-8] process has died [pid 19766, exit code 1, cmd /home/user11/catkin_ws/src/turtlebot3_autorace_2020/turtlebot3_autorace_parking/turtlebot3_autorace_parking_detect/nodes/detect_parking /detect/image_input:=/camera/image_compensated /detect/image_input/compressed:=/camera/image_compensated/compressed /detect/scan:=/scan /detect/image_output:=/detect/image_parking /detect/image_output/compressed:=/detect/image_parking/compressed __name:=detect_parking __log:=/home/user11/.ros/log/a39aa3ce-3523-11eb-80e3-e0d4e8f1883a/detect_parking-8.log]. log file: /home/user11/.ros/log/a39aa3ce-3523-11eb-80e3-e0d4e8f1883a/detect_parking-8*.log

chang556, 2020-12-03:
請問為什麼原本執行成功,後面反而會發生這問題

BENBEY-I, 2022-06-03:
Hello, I have the same problem, have you managed to find the solution
[image]
?

chang556, 2022-06-22:
[image: image.png]

chang556, 2022-06-22:
[image]

BENBEY-I, 2022-06-22:
hi, i think the openCV wasn't installed, 
this solved it 
`pip3 install opencv-contrib-python`
`pip3 install opencv-python`
you can see the original answer I got in ROS ANSWER 
[](https://answers.ros.org/question/401862/turtlebot3-traffic-sign-detection-problem/)

closed 2025-02-24 by sunghowoo