ROBOTIS-GIT/turtlebot3_autorace #58: Detect Lane has no attribute 'left_fit'

lucyivt, 2024-03-11:
Hi, I've been trying to get the lane detection code to work but keep getting this error: 

AttributeError: 'DetectLane' object has no attribute 'left_fit'

The problem arises in line number 159 of detect_lane.py: 
`if yellow_fraction > 3000:`
`       self.left_fitx, self.left_fit = self.fit_from_lines(self.left_fit , cv_yellow_lane)`
 `       self.mov_avg_left = np.append(self.mov_avg_left,np.array([self.left_fit]), axis=0)`

Where we try to initiate self.left_fit while also using it in the function parameter. I've been struggling to figure out how to fix this problem, I've tried to initiate self.left_fit before this line (making it an array) but more errors occur. (I put the turtlebot between the yellow and white lines.) Any help would be greatly appreciated!

GyuH13, 2025-03-11:
Hello, @lucyivt 
I apologize for the late response. 
The Noetic version is no longer being released. The AutoRace has been updated to the Humble version. Please check the Humble version.

GyuH13, 2025-03-18:
Since there are no further questions, I will close this issue.

closed 2025-03-18 by GyuH13