ROBOTIS-GIT/turtlebot3_autorace #62: Autorace no images

ZaynabEM, 2025-03-26:
I am following the <a href="https://emanual.robotis.com/docs/en/platform/turtlebot3/autonomous_driving/">autorace tutorial</a>, in ROS2 humble distribution.
I verified that the clone was done on the humble branch.
Verified the dependencies : 
```bash
libopencv-dev is already the newest version (4.5.4+dfsg-9ubuntu4).
python3-opencv is already the newest version (4.5.4+dfsg-9ubuntu4).
ros-humble-cv-bridge is already the newest version (3.2.1-1jammy.20241128.024139).
ros-humble-image-pipeline is already the newest version (3.0.6-1jammy.20250121.134247).
ros-humble-image-transport is already the newest version (3.1.10-1jammy.20241128.022756).
ros-humble-vision-opencv is already the newest version (3.2.1-1jammy.20241128.025550).
```
I then launched the `intrinsic_camera_calibration.launch.py` and got the following : 
```bash
[INFO] [launch]: All log files can be found below /home/user/.ros/log/...
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [republish-1]: process started with pid [137887]
[INFO] [component_container-2]: process started with pid [137889]
[component_container-2] [INFO] [1742983989.286426635] [camera.image_proc_container]: Load Library: /opt/ros/humble/lib/librectify.so
[component_container-2] [INFO] [1742983989.349730415] [camera.image_proc_container]: Found class: rclcpp_components::NodeFactoryTemplate<image_proc::RectifyNode>
[component_container-2] [INFO] [1742983989.349778483] [camera.image_proc_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<image_proc::RectifyNode>
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/camera/rectify_node' in container '/camera/image_proc_container'
[component_container-2] [INFO] [1742983989.369833915] [camera.image_proc_container]: Load Library: /opt/ros/humble/lib/libdebayer.so
[component_container-2] [INFO] [1742983989.371426023] [camera.image_proc_container]: Found class: rclcpp_components::NodeFactoryTemplate<image_proc::DebayerNode>
[component_container-2] [INFO] [1742983989.371443194] [camera.image_proc_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<image_proc::DebayerNode>
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/camera/debayer_node' in container '/camera/image_proc_container'
```

Then launched the `extrinsic_camera_calibration.launch.py` with the indicated parameter `calibration_mode:=True`, and got the following:
```bash
[INFO] [launch]: All log files can be found below /home/user/.ros/log/...
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [image_projection-1]: process started with pid [138009]
[INFO] [image_compensation-2]: process started with pid [138011]
```

When I checked the messages `/camera/image_extrinsic_calib` and `/camera/image_projected` in rqt_image_view, nothing is being published.

I got the calibration file in the `projection.yaml`:
```yaml
camera:
  image_projection:
    ros__parameters:
      camera:
        extrinsic_camera_calibration:
          top_x: 86
          top_y: 0
          bottom_x: 169
          bottom_y: 120
```

and the `compensation.yaml`:
```yaml
camera:
  image_compensation_projection:
    ros__parameters:
      camera:
        extrinsic_camera_calibration:
          clip_hist_percent: 1.0
```

So I tried to move on with the tutorial, and launched `detect_lane.launch.py` in the `turtlebot3_autorace_detect` instead of in `turtlebot3_autorace_camera` as it is not there, and then `control_lane.launch.py` in the `turtlebot3_autorace_mission` instead of `turtlebot3_autorace_driving` again because it was not there.

Here is a list of the topics that I have:
```bash
/camera/image_projected
/clock
/cmd_vel
/imu
/joint_states
/odom
/parameter_events
/performance_metrics
/robot_description
/rosout
/scan
/tf
/tf_static
```
Still the robot is not giving any images on the topic `/camera/image_projected` and not doing anything.
I then followed the video <a href="https://www.youtube.com/watch?v=DhSZo3dGW6A&ab_channel=ROBOTISOpenSourceTeam">TurtleBot3 71 Autorace Traffic Sign Detection</a> but still nothing.

What am I missing?

chan-1207, 2025-04-09:
Hello,

I’ve checked out and found a couple of issues:

1. First, please set the model of your turtlebot to buger_cam.
`export TURTLEBOT3_MODEL=burger_cam`

1. please `run ros2 run turtlebot3_autorace_camera intrinsic_camera_calibration.launch.py` and note the topic name being published via rqt_graph.

1. In `/turtlebot3_autorace/turtlebot3_autorace_camera/launch/extrinsic_camera_calibration.launch.py`, replace remapping with the topic name that is being issued.

In the example photo below, the camera topic name is /camera/image_raw/compressed, so we have remapped it accordingly.

[image]
[image]

GyuH13, 2025-04-09:
@ZaynabEM Hello
Please show me the rqt graph
If you operate usually, below topic will be shown.
```
/camera/camera_info
/camera/image_projected
/camera/image_raw
/camera/image_raw/compressed
```
I think you should `export TURTLEBOT3_MODEL=burger_cam`

ZaynabEM, 2025-04-09:
Hello, 

Thank you so much @chan-1207 and @GyuH13 for the modifications and indications, I just verified and indeed now I have all the topics and parameters.
I have also verified all the missions available and they work fine.

Just needed to fix some versions because the `rosdep` and therefore the `colcon build` were producing the following error : 
```bash
CMake Error at CMakeLists.txt:24 (find_package):
  By not providing "Findgazebo_ros_pkgs.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "gazebo_ros_pkgs", but CMake did not find one.

  Could not find a package configuration file provided by "gazebo_ros_pkgs"
  with any of the following names:

    gazebo_ros_pkgsConfig.cmake
    gazebo_ros_pkgs-config.cmake
```

The reason was the version mismatch preventing the installation of the `gazebo_ros_pkgs` , which was fixed after reinstalling the `gz-tools2`:
```bash
 gz-tools2 : Conflicts: gazebo (>= 11.0.0) but 11.10.2+dfsg-1 is to be installed
             Conflicts: gazebo (<= 11.14.0) but 11.10.2+dfsg-1 is to be installed
```

After that, everything worked fine.

Thanks again for your help!
I will be closing the issue as completed.

closed 2025-04-09 by ZaynabEM

GyuH13, 2025-04-10:
@ZaynabEM 
I'm glad to hear that the issue has been resolved!
If you encounter any inconvenience or bugs while using it, please feel free to let us know anytime.
Your contribution helps us make a better TurtleBot3.
Thank you!